STMicroelectronics/st-mems-android-linux-drivers-iio #3: compilation of st_lis2dw12_buffer.c with kernel 4.15 fails

MarkusDost, 2022-10-05:
I'm using the Kernel 4.15. and get the following error when I try to compile the driver for the LIS2DW12/IIS2DLPC:

st_lis2dw12_buffer.c:308:18: error: implicit declaration of function 'devm_iio_kfifo_allocate'; did you mean 'iio_kfifo_allocate'? [-Werror=implicit-function-declaration]

The function 'devm_iio_kfifo_allocate' was changed to a private function in kernel 5.13 and replaced by 'devm_iio_kfifo_buffer_setup' in header 'kfifo_buf.h'

mariotesi, 2022-10-05:
4.15 or 5.15 ?

MarkusDost, 2022-10-05:
Sorry my mistake. I mean 5.15

ciocca-d, 2022-10-05:
commit 3925f9892341e3abd86488717e5238135a899f8c pushed out few hours ago should fix the issue you are facing.

closed 2022-10-05 by ciocca-d

MarkusDost, 2022-10-06:
Thank you, this fix works.