STMicroelectronics/st-mems-android-linux-drivers-iio #5: LSM6DSO iio errors

rosterloh, 2022-11-04:
I have added your drivers to the current released tag of the [Raspberry Pi kernel](https://github.com/raspberrypi/linux/tree/1.20221028) version 5.15.74 and compiled the modules for st_lsm6dsox. The module is loaded with the following overlay.

```
/dts-v1/;
/plugin/;
#include <dt-bindings/interrupt-controller/irq.h>

/ {
        compatible = "brcm,bcm2835";

        fragment@0 {
                target = <&i2c1>;
                __overlay__ {
                        #address-cells = <1>;
                        #size-cells = <0>;
                        status = "okay";
                        clock-frequency = <400000>;

                        lsm6dso: lsm6dso@6a {
                                compatible = "st,lsm6dso";
                                reg = <0x6a>;
                                interrupt-parent = <&gpio>;
                                interrupts = <22 IRQ_TYPE_EDGE_FALLING>,
					     <27 IRQ_TYPE_EDGE_FALLING>;
				st,int-pin = <1>;
                        };
                };
        };

};
```

The following failures are then observed in dmesg

[    8.657920] st_lsm6dsox: disagrees about version of symbol iio_push_event
[    8.657955] st_lsm6dsox: Unknown symbol iio_push_event (err -22)
[    8.657981] st_lsm6dsox: disagrees about version of symbol iio_push_to_buffers
[    8.657994] st_lsm6dsox: Unknown symbol iio_push_to_buffers (err -22)
[    8.658021] st_lsm6dsox: disagrees about version of symbol iio_device_release_direct_mode
[    8.658033] st_lsm6dsox: Unknown symbol iio_device_release_direct_mode (err -22)
[    8.658073] st_lsm6dsox: disagrees about version of symbol devm_iio_kfifo_buffer_setup_ext
[    8.658085] st_lsm6dsox: Unknown symbol devm_iio_kfifo_buffer_setup_ext (err -22)
[    8.658117] st_lsm6dsox: disagrees about version of symbol iio_device_alloc
[    8.658129] st_lsm6dsox: Unknown symbol iio_device_alloc (err -22)
[    8.658153] st_lsm6dsox: disagrees about version of symbol iio_device_free
[    8.658164] st_lsm6dsox: Unknown symbol iio_device_free (err -22)
[    8.658206] st_lsm6dsox: disagrees about version of symbol iio_show_mount_matrix
[    8.658217] st_lsm6dsox: Unknown symbol iio_show_mount_matrix (err -22)
[    8.658235] st_lsm6dsox: disagrees about version of symbol iio_device_claim_direct_mode
[    8.658246] st_lsm6dsox: Unknown symbol iio_device_claim_direct_mode (err -22)
[    8.658284] st_lsm6dsox: disagrees about version of symbol __iio_device_register
[    8.658296] st_lsm6dsox: Unknown symbol __iio_device_register (err -22)
[    8.658330] st_lsm6dsox: disagrees about version of symbol __devm_iio_device_register
[    8.658341] st_lsm6dsox: Unknown symbol __devm_iio_device_register (err -22)
[    8.658372] st_lsm6dsox: disagrees about version of symbol devm_iio_device_alloc
[    8.658383] st_lsm6dsox: Unknown symbol devm_iio_device_alloc (err -22)
[    8.658399] st_lsm6dsox: disagrees about version of symbol iio_get_time_ns
[    8.658410] st_lsm6dsox: Unknown symbol iio_get_time_ns (err -22)
[    8.658426] st_lsm6dsox: disagrees about version of symbol iio_device_unregister
[    8.658437] st_lsm6dsox: Unknown symbol iio_device_unregister (err -22)

mariotesi, 2022-11-04:
Hi,

It seems there are some missing kernel modules loaded on your target board.
Please consider the lsm6dsox sensor module depends on the IIO framework so also the IIO modules must be installed on the board. I suggest to repeat the module_install after enabling the lsm6dsox module in the menuconfig which automatically selects the build of all the dependent modules as well.

On my RPI with k5.15 this is the modules dependency after command  lsmod | grep lsm6dso:
st_lsm6dsox_spi        16384  0
regmap_spi             16384  1 st_lsm6dsox_spi
st_lsm6dsox_i2c        16384  0
st_lsm6dsox            57344  2 st_lsm6dsox_i2c,st_lsm6dsox_spi
kfifo_buf              16384  1 st_lsm6dsox
industrialio           81920  2 st_lsm6dsox,kfifo_buf
regmap_i2c             16384  1 st_lsm6dsox_i2c

rosterloh, 2022-11-06:
Hi @mariotesi, thanks for your response. The output of my lsmod shows all the required modules are loaded on my system.
```
Module                  Size  Used by
regmap_spi             16384  0
kfifo_buf              16384  0
industrialio           90112  1 kfifo_buf
regmap_i2c             16384  2 lm75,rtc_pcf85063
```
I cannot however successfully load the st_lsm6dsox module. Can you think of any other things to try?

closed 2024-02-09 by rosterloh