STMicroelectronics/st-mems-android-linux-drivers-iio #7: Slow when keep read value?

WolsYang, 2023-02-17:
I got really slowy speed when read iio node, and some value event not correct.
Is any way I can fix it??
Here is my script and output.
```
echo "Hello iio test"
cat /sys/bus/iio/devices/iio\:device2/sampling_frequency 
while :
do
        cat /sys/bus/iio/devices/iio:device2/in_accel_x_raw >> /dev/kmsg
        cat /sys/bus/iio/devices/iio:device2/in_accel_y_raw >> /dev/kmsg
        cat /sys/bus/iio/devices/iio:device2/in_accel_z_raw >> /dev/kmsg
        cat /sys/bus/iio/devices/iio:device1/in_anglvel_x_raw >> /dev/kmsg
        cat /sys/bus/iio/devices/iio:device1/in_anglvel_y_raw >> /dev/kmsg
        cat /sys/bus/iio/devices/iio:device1/in_anglvel_z_raw >> /dev/kmsg
    sleep 0.01
done
```

```
root@iBox-500:~/kivyT# sh test.sh                                                                                                                                                                              
Hello iio test                                                                                                                                                                                                 
833.000000                                                                                                                                                                                                     
[ 1131.604427] 1548                                                                                                                                                                                            
[ 1131.616635] -368                                                                                                                                                                                            
[ 1131.628900] -16176                                                                                                                                                                                          
[ 1131.956560] 9                                                                                                                                                                                               
[ 1132.283833] -70                                                                                                                                                                                             
[ 1132.558214] -68                                                                                                                                                                                             
[ 1132.584233] 1486                                                                                                                                                                                            
[ 1132.596522] -408                                                                                                                                                                                            
[ 1132.608636] -16095                                                                                                                                                                                          
[ 1132.935723] 7                                                                                                                                                                                               
[ 1133.262978] -70                                                                                                                                                                                             
[ 1133.590466] -67                                                                                                                                                                                             
[ 1133.616815] 1444                                                                                                                                                                                            
[ 1133.628973] -423                                                                                                                                                                                            
[ 1133.641130] -16098                                                                                                                                                                                          
^C                                                                                                                                                                                                             
root@iBox-500:~/kivyT# 
```

rosterloh, 2024-02-09:
I have the same issue with the LSM6DSO. @WolsYang did you ever find a solution to this?

WolsYang, 2024-02-09:
@rosterloh 
No, we don't fix it.
It was a long time ago.
The last thing I remember I found it is slow because the driver used the delay to avoid reading too fast.
But I don't know the reason, and forgot where I found it.

mariotesi, 2024-02-21:
Hi [WolsYang](https://github.com/WolsYang)

the [delay](https://github.com/STMicroelectronics/st-mems-android-linux-drivers-iio/blob/master/drivers/iio/stm/imu/st_lsm6dsrx/st_lsm6dsrx_core.c#L753) introduced in the driver, only for the single reading of the raw data, has the purpose of giving the sensor time to stabilize after it has been enabled, this is related to the settling time of the filters. The number of samples to discard after sensor activation is reported in application note AN5358 and depends on the ODR of the selected sensor.
This delay can be reduced by setting high ODR.
Anyway if the aim is to read data at the selected ODRs, we suggest using the IIO buffer functionality, abandoning the raw reading approach of individual data axis which by its nature requires many more operations and is therefore slower (an example code for reading IIO sensor devices using IIO buffer is [generic_buffer](https://github.com/torvalds/linux/blob/master/tools/iio/iio_generic_buffer.c))

closed 2024-05-03 by mariotesi