STMicroelectronics/st-mems-android-linux-drivers-iio #18: Power up fail for lis2dux12/lis2duxs12

lorenzodallacasa, 2024-05-14:
Hi,
I'm using LIS2DUX12 accelerometer on a custom PCB with SPI interface in a Linux environment and I was experiencing a failure during the probe of the device with error message `[    0.561627] st_lis2duxs12_spi spi0.0: unsupported whoami [ff]`
Reading the datasheet documentation I found out that 

> When Vdd and Vdd_IO are set, the device enters a deep power-down state

So I read how to exit from a deep power-down state

> If the SPI interface is used, the LIS2DUX12 can move from deep power-down to soft power-down by writing the
bit SOFT_PD of EN_DEVICE_CONFIG (3Eh) to 1

and took a look into the implementation of the function `st_lis2duxs12_power_up_command(struct st_lis2duxs12_hw *hw)` inside the file 
*st_lis2duxs12_core.c*

To my surprise I found out that the code does not implement what's described in the documentation but instead was reading WHO_AM_I (0Fh) register. 

https://github.com/STMicroelectronics/st-mems-android-linux-drivers-iio/blob/d407e7b289169e5018ee8eefa774bc94fcc53073/drivers/iio/stm/accel/st_lis2duxs12_core.c#L313

To this SPI transaction the accelerometer does not respond and the MISO line stays high.

So, I tried to write EN_DEVICE_CONFIG (3Eh) to 1 instead and by doing so the probe returned with success!
I did this by putting the magic numer 0x3E as an argument to the `regmap_write()` function because there is no definition of that register address inside `st_lis2duxs12.h` file. Does this file, as well as others related to LIS2DUX12, need to be updated to reflect the latest changes with this device/documentation?

Thanks

mariotesi, 2024-05-16:
Hi Lorenzo,

You're absolutely right, we missed the driver update on the last revision of the documentation.
The patch will soon be available on public repo.

lorenzodallacasa, 2024-05-16:
Ok nice! Thank you very much

closed 2024-05-27 by lorenzodallacasa