STMicroelectronics/st-mems-android-linux-drivers-iio #24: LSM303AGR Magnetometer interrupt not enabled by driver

rbenet-hp, 2025-02-27:
Hi,
I am working on a system with LSM303AGR over I2C, on Linux Kernel 5.15.

We are trying to use the magnetomer using the IIO drivers on this repository. (we are also using the accelerometer, and that is working just fine)

When we issue the following commands on the shell, we get no output from the device:
```bash
MAGP=$(dirname $(find /sys/bus/iio/devices/iio\:device*/ -name "in_magn_x_raw" -print -quit)) ;
MAGD="/dev/$(basename $MAGP)";
echo 0 > "$MAGP/buffer/enable";
echo 1 > "$MAGP/scan_elements/in_magn_x_en";
echo 1 > "$MAGP/scan_elements/in_magn_y_en";
echo 1 > "$MAGP/scan_elements/in_magn_z_en";
echo 1 > "$MAGP/scan_elements/in_timestamp_en";
echo 1 > "$MAGP/buffer/enable";
echo $MAGD
dd if="$MAGD" bs=16 count=160 status=none | xxd
```

Many things could be wrong at this point (Hardware, GPIO configuration, I2C bus,...), *but* note that if before the code above, we just do:
`i2cset -f -y 5 0x1e 0x62 0x50 bb`    (which enables bit `INT_MAG_PIN` on register `CFG_REG_C_M` - note it also sets BDU, as it seems to be On by default and we just aim to keep as it is)

After adding the above, the `dd` command is sucessful:

```bash
# dd if="$MAGD" bs=16 count=160 status=none | xxd
00000000: 72ef dc3c 7d07 ffff 20bd cec9 6b22 ea16  r..<}... ...k"..
00000010: 86ef 7b3c 2c07 0000 e0b3 dbc9 6b22 ea16  ..{<,.......k"..
00000020: 8cef 7b3c 9806 0000 786f 1ace 6b22 ea16  ..{<....xo..k"..
00000030: 86ef 8c3c f906 0000 c886 21ce 6b22 ea16  ...<......!.k"..
00000040: 86ef 8a3c fd07 ffff 540f 3fce 6b22 ea16  ...<....T.?.k"..
00000050: 88ef 8f3c 7206 ffff bc49 4cce 6b22 ea16  ...<r....IL.k"..
00000060: 8cef 773c 7606 ffff 003f 59ce 6b22 ea16  ..w<v....?Y.k"..
00000070: 85ef 7b3c 8307 ffff 1017 71ce 6b22 ea16  ..{<......q.k"..
00000080: 87ef 713c 8707 0000 2c25 7ece 6b22 ea16  ..q<....,%~.k"..
00000090: 87ef 6a3c 5007 ffff 948b 8cce 6b22 ea16  ..j<P.......k"..
...
```

Looking at the code, it seems like it is setting `CFG_REG_C_M.INT_MAG` in https://github.com/STMicroelectronics/st-mems-android-linux-drivers-iio/blob/621b086d888f4486d72a630f2123a6985c34054d/drivers/iio/stm/magnetometer/st_mag40_buffer.c#L137 , but it is not setting `INT_MAG_PIN`.

Note that for further confirmation, I have also looked with `irqtop` to see whether indeed the issue is the interrupt not being generated.
Before my change, the IRQ associated with the magnetometer is quiet.
After my change, I can see the count of IRQ increasing.

Might I be missing something in terms of configuration/setup/DTS/... ? 
Or is this just a plain bug in the driver ?

Thx,
Raul

mariotesi, 2025-02-28:
Hi Raul,

Assuming you want to use data ready signal as interrupt source (not the interrupt threshold) to have this configuration there is not reason to set to 1 the INT_MAG_PIN bit (reg  CFG_REG_C_M). Leave it at 0. 

[image]

INT_MAG_PIN bit purpose is to divert threshold interrupt signal to the INT_MAG/DRDY pin (7).  In our hypothesis this is out of your interest, so leave it clear. 

Please refers to the following link, capters 3.3.3, 3.3.4 and 3.6.1 in [AN4825](https://www.st.com/resource/en/application_note/an4825-ultracompact-highperformance-ecompass-module-based-on-the-lsm303agr-stmicroelectronics.pdf)

 Likely, the reason why the driver does not work could be found in the interrupt configuration of your board. What is your platform configuration for the drdy interrupt on your device tree ? Is it edge or level triggered ? What about the interrupt polarity ?
 
Furthermore I suggest you to use more specific applications to read data from the IIO sensor. For example, [generic_buffer](https://github.com/torvalds/linux/blob/master/tools/iio/iio_generic_buffer.c) or the tool suite [libiio](https://github.com/analogdevicesinc/libiio) (both are open source projects) that automatically implements all the operations you have collected in your bash script.

rbenet-hp, 2025-03-04:
Mario, 
thank your for the prompt response.

With your guidance and insistance on the correctness of the driver, I managed to get to the bottom of my issue.
It turns out the ST IIO driver for the magnetometer in our repo had been committed (by a 3rd party company) with the IRQ polarity inverted, without any warning or comment or specific commit. Go figure !

After restoring the driver to its original code, things are working as expected.

Thanks again.

(for the avoidance of doubt - the code above (shell script) is only used for debugging/exploring - for production we do read the IIO device file directly from C code. We decided not to use `libiio` because it felt like "too much" for our simple very constrained use case).