STMicroelectronics/st-mems-android-linux-drivers-iio #29: lsm6dso FIFO bulk read through iio_generic_buffer

DBert88, 2025-05-29:
We are currently not able to perform a bulk fifo read of the lsm6dso accelerometer. The device is connected through SPI and the dts entry look like this: 

                        spi0: spi@f8000000 {
				pinctrl-names = "default";
				pinctrl-0 = <&pinctrl_spi0_default>;
				status = "okay";
				num-cs = <1>;
				cs-gpios = <&pioA PIN_PA17 GPIO_ACTIVE_LOW>, <0>, <0>, <0>;
				status = "okay";

                          lsm6dsox-imu@0 {
					compatible = "st,lsm6dso";
					spi-max-frequency = <1000000>;
					reg = <0x0>;
					spi-cs-high;   /* cs reversed due to the transistor */	
					interrupt-parent = <&pioBU>;
					interrupts = <2 IRQ_TYPE_EDGE_RISING>;
					st,int-pin = <1>;
                          };

The problem is that no buffer nor scan_elements entry is shown at userspace. 
Trying to debug what was happening, we found that from the spi probe function of the lsm6dsox driver `static int sts_lsm6dsox_spi_probe(struct spi_device *spi)` after calling `devm_regmap_init_spi(...)` we get a spi_device "spi" with `spi->irq == 0` and then no buffer setup is performed from the "main" probe function of the driver. 
What can be wrong?

mariotesi, 2025-07-10:
Hi DBert88,

Have you checked if there are any specific messages in the kernel log regarding interrupt configuration?

DBert88, 2025-07-23:
Hi @mariotesi we just found that we were trying to use a non-interrupt controller (pioBU in the dts snippet), we will try to use a different gpio-chip. Shouldn't we anyway be able to perform a bulk read even without the FIFO interrupt configuration, in a trigger-less fashion?

mariotesi, 2025-07-29:
Hi DBert88,

The lsm6dsox IIO driver supports external software trigger, so you can use an IIO hrtimer to trigger the sensor reading. For example, configuring the kernel to export the iio-trig-hrtimer module and enable configfs, it is possible to create an IIO hrtimer as a trigger source for the sensor driver:

-  mount configfs and add sw trigger support to the board
$ mkdir /config
$ modprobe iio-trig-hrtimer
$ mount -t configfs none /config
$ mkdir /config/iio/triggers/hrtimer/hr_sw_trigger_timer

- set hr_sw_trigger_timer odr to 10 Hz
$ echo 10 > /sys/bus/iio/devices/trigger0/sampling_frequency

- using [generic_buffer](https://github.com/torvalds/linux/tree/master/tools/iio) (or manually writing the trigger name in the sys/bus/iio/devicesiio:device<x>/trigger/current_trigger of the lsm6dso) it is possible to read the sensor data:
$ generic_buffer -n lsm6dso_accel -a -t hr_sw_trigger_timer -c 10
$ generic_buffer -n lsm6dso_gyro -a -t hr_sw_trigger_timer -c 10
$ generic_buffer -n lsm6dso_magn -a -t hr_sw_trigger_timer -c 10