STMicroelectronics/st-mems-android-linux-drivers-iio #30: Incompatible with ASM330LHH and ASM330LHM

1plus1overtake2, 2025-07-24:
In FIFO_CTRL3 (09h)
BDR_XL_[3:0]

There is no 1011: 1.6 Hz option in ASM330LHH/M

1plus1overtake2, 2025-07-24:
when I run the linux iio tool iio_generic_buffer, cannot enable the buffer, 
/tmp # ./iio_generic_buffer -c 10 -n asm330lhh_accel -g -a
iio device number being used is 3
trigger-less mode selected
Enabling all channels
Enabling: in_accel_x_en
Enabling: in_count_en
Enabling: in_accel_z_en
Enabling: in_timestamp_en
Enabling: in_accel_y_en
Failed to enable buffer '/sys/bus/iio/devices/iio:device3/buffer0': Invalid argument
Disabling: in_accel_x_en
Disabling: in_count_en
Disabling: in_accel_z_en
Disabling: in_timestamp_en
Disabling: in_accel_y_en

mariotesi, 2025-07-29:
Hi [1plus1overtake2](https://github.com/1plus1overtake2),

Could you verify that the interrupt you're passing to the driver via the device tree is correct? Try looking at the output of the shell command "cat /proc/interrupts | grep asm330"